OpenHUTB/hutb #18: 是不是在蓝色加号那里添加贡献者列表？

lujianjjjj, 2024-09-03:
[image]

Candela-best, 2024-09-03:
这个在哪个文件下啊 没找到

donghaiwang, 2024-09-03:
> ![e02dbc1628aaec8e7e8575b19c75544](https://private-user-images.githubusercontent.com/170411192/363893601-0dacd9bb-aa2b-4a53-b0e1-2c4a0c7bab95.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNTcwNDYsIm5iZiI6MTcyNTM1Njc0NiwicGF0aCI6Ii8xNzA0MTExOTIvMzYzODkzNjAxLTBkYWNkOWJiLWFhMmItNGE1My1iMGUxLTJjNGEwYzdiYWI5NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QwOTQ1NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZGJmMTNkNzdlYWEzNjY1OGE0NDgyOTM4NGQ2MmQ2MWE0MjQzOGFjODFlOWY1Y2ExN2RiMTQ1M2NkMjM3ZWQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.aKShVsGgvfK1d0JbMnFwrFjb7wtiPnracO-lo6K2Mw8)

是的，蓝色是修改后的效果。

donghaiwang, 2024-09-03:
> 这个在哪个文件下啊 没找到

是修改 [这个文件](https://github.com/OpenHUTB/carla_cpp/blob/dev/README.md)

closed 2025-02-09 by donghaiwang