OpenHUTB/hutb #19: 虚拟机具体怎么安装？

lujianjjjj, 2024-09-03:
[image]

donghaiwang, 2024-09-03:
先安装 VMware 或者 virtual box 虚拟机软件，然后下载这里的虚拟机文件，用虚拟机软件打开虚拟机文件。

lujianjjjj, 2024-09-03:
这个吗？
[image]

lujianjjjj, 2024-09-03:
那些虚拟文件全部都要下载吗？

lujianjjjj, 2024-09-03:
[image]
这个行吗

donghaiwang, 2024-09-03:
> 那些虚拟文件全部都要下载吗？

只需要下载需要的，我们课程是ROS Kinetic版本

donghaiwang, 2024-09-03:
> ![33398479610e9baaef6bb24817480c9](https://private-user-images.githubusercontent.com/170411192/363941868-a46ecfd9-2fa8-44ec-bd7f-bc8162e5a1e4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjA3MDYsIm5iZiI6MTcyNTM2MDQwNiwicGF0aCI6Ii8xNzA0MTExOTIvMzYzOTQxODY4LWE0NmVjZmQ5LTJmYTgtNDRlYy1iZDdmLWJjODE2MmU1YTFlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMDQ2NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZTk0NTZiZTljYTE5YTdmODQzMDg2MjUxYmZmNDJjMTUwODAzZWRlZGM3MzEwZWZhZDUxYjQzODA5ZjljYjQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.flap0T2uwngtCpkIFUz0EnTkvzC9b3zVewUu_KJ1Wro) 这个行吗

这个只有虚拟机部分功能，没试过，可以试试。

lujianjjjj, 2024-09-03:
虚拟机是直接在VMware上下载安装吗

lujianjjjj, 2024-09-03:
VMware虚拟机是直接在VMware官网上下载安装吗

donghaiwang, 2024-09-03:
> VMware虚拟机是直接在VMware官网上下载安装吗

Vmware虚拟机软件搜索就可以找到安装包，然后再装Ubuntu虚拟机，这里提供的是一种安装好ROS的Ubuntu虚拟机 https://rosi-images.d10.us.swri.org/

closed 2025-02-09 by donghaiwang