OpenHUTB/hutb #27: 不会修改，看不懂

Candela-best, 2024-09-03:
[image]
这里不会就操作了，看不懂，该怎么修改
[image]

donghaiwang, 2024-09-03:
进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。

Candela-best, 2024-09-03:
> 进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。

[image]
就是这样的，两个文件也找得到，但是该怎么修改

Candela-best, 2024-09-03:
后面的看不懂了。。。

donghaiwang, 2024-09-03:
> > 进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。
> 
> ![屏幕截图 2024-09-03 184343](https://private-user-images.githubusercontent.com/180130297/363950770-009f93a5-ade4-4031-8e96-57494f9f8a59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjA3NTksIm5iZiI6MTcyNTM2MDQ1OSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTUwNzcwLTAwOWY5M2E1LWFkZTQtNDAzMS04ZTk2LTU3NDk0ZjlmOGE1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMDQ3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZjBhMDZkYmFhZDc3ZWZlODk2MTM0ODBkYmFhZGFmZTg4YWYxNDNhMjc2OWM1YTZjMzg4YzdiNTRlMjQ3M2RkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XgYP56ef7mW6a6uqUSmz5ygiXr-66d7ExjuAo0MC_V8) 就是这样的，两个文件也找得到，但是该怎么修改

不是修改这里，这是生成代码的文档，我们需要到代码目录里的.cpp和.h文件中去修改和增加注释

donghaiwang, 2024-09-03:
> > > > 进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。
> > > 
> > > 
> > > ![屏幕截图 2024-09-03 184343](https://private-user-images.githubusercontent.com/180130297/363950770-009f93a5-ade4-4031-8e96-57494f9f8a59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjA3NTksIm5iZiI6MTcyNTM2MDQ1OSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTUwNzcwLTAwOWY5M2E1LWFkZTQtNDAzMS04ZTk2LTU3NDk0ZjlmOGE1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMDQ3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZjBhMDZkYmFhZDc3ZWZlODk2MTM0ODBkYmFhZGFmZTg4YWYxNDNhMjc2OWM1YTZjMzg4YzdiNTRlMjQ3M2RkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XgYP56ef7mW6a6uqUSmz5ygiXr-66d7ExjuAo0MC_V8) 就是这样的，两个文件也找得到，但是该怎么修改
> > 
> > 
> > 不是修改这里，这是生成代码的文档，只是打开看一看，我们需要到代码目录里的.cpp和.h文件中去修改和增加注释
> 
> 没有找到代码目录。。。没有.p和.cpp的文件。。。

在LibCarla和CarlaUE4目录下

Candela-best, 2024-09-03:
> 不是修改这里，这是生成代码的文档，只是打开看一看，我们需要到代码目录里的.cpp和.h文件中去修改和增加注释

我用VS2022打开这个MD文件之后，是这个吗
[image]
[image]

donghaiwang, 2024-09-03:
> > > > 进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。
> > > 
> > > 
> > > ![屏幕截图 2024-09-03 184343](https://private-user-images.githubusercontent.com/180130297/363950770-009f93a5-ade4-4031-8e96-57494f9f8a59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjA3NTksIm5iZiI6MTcyNTM2MDQ1OSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTUwNzcwLTAwOWY5M2E1LWFkZTQtNDAzMS04ZTk2LTU3NDk0ZjlmOGE1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMDQ3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZjBhMDZkYmFhZDc3ZWZlODk2MTM0ODBkYmFhZGFmZTg4YWYxNDNhMjc2OWM1YTZjMzg4YzdiNTRlMjQ3M2RkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XgYP56ef7mW6a6uqUSmz5ygiXr-66d7ExjuAo0MC_V8) 就是这样的，两个文件也找得到，但是该怎么修改
> > 
> > 
> > 不是修改这里，这是生成代码的文档，只是打开看一看，我们需要到代码目录里的.cpp和.h文件中去修改和增加注释
> 
> 我用VS2022打开这个MD文件之后，是这个吗 ![屏幕截图 2024-09-03 185926](https://private-user-images.githubusercontent.com/180130297/363955330-6b57211d-3861-46ef-b0ae-e494ff713fda.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjE1ODYsIm5iZiI6MTcyNTM2MTI4NiwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU1MzMwLTZiNTcyMTFkLTM4NjEtNDZlZi1iMGFlLWU0OTRmZjcxM2ZkYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYWZjNGZjNDIxMGQyZThkNzQ1MzllZDM3ZDhkZGYwMmE2ZDhlNDg1NDE1N2MwZmVjZjgzOWM3NTE3MWQ5OWQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XqtwvzgW0TTRAKQRfE1AFvDQl0-TolS0HLvvzgjo7w8) ![屏幕截图 2024-09-03 185841](https://private-user-images.githubusercontent.com/180130297/363955474-36650e59-d705-46b9-a130-d80044742fa8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjE1ODYsIm5iZiI6MTcyNTM2MTI4NiwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU1NDc0LTM2NjUwZTU5LWQ3MDUtNDZiOS1hMTMwLWQ4MDA0NDc0MmZhOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MzUxYjZmZGVhZDVmOWJhNzM4MDJiMjI0MmY5ZThjZWIzMDI1YzJjNzI2Yjg2ZGU4MmZjZTdjODEwNTZjODZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qyIJUq1QvZ5wg3k9mVMhAhzHKIW60VznpRRIrWQW0X8)

这个是修改项目说明，比如增加贡献者名

Candela-best, 2024-09-03:
> 这个是修改项目说明，比如增加贡献者名

[image]
不是要增加贡献者名吗，我把我的打上去了，emmm接下来该怎么增加上去。。。
[image]

donghaiwang, 2024-09-03:
> > > > > > 进入`carla_cpp\Doxygen\html`目录打开对应的html文件查看生成的文档效果。
> > > > > 
> > > > > 
> > > > > ![屏幕截图 2024-09-03 184343](https://private-user-images.githubusercontent.com/180130297/363950770-009f93a5-ade4-4031-8e96-57494f9f8a59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjA3NTksIm5iZiI6MTcyNTM2MDQ1OSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTUwNzcwLTAwOWY5M2E1LWFkZTQtNDAzMS04ZTk2LTU3NDk0ZjlmOGE1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMDQ3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZjBhMDZkYmFhZDc3ZWZlODk2MTM0ODBkYmFhZGFmZTg4YWYxNDNhMjc2OWM1YTZjMzg4YzdiNTRlMjQ3M2RkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XgYP56ef7mW6a6uqUSmz5ygiXr-66d7ExjuAo0MC_V8) 就是这样的，两个文件也找得到，但是该怎么修改
> > > > 
> > > > 
> > > > 不是修改这里，这是生成代码的文档，只是打开看一看，我们需要到代码目录里的.cpp和.h文件中去修改和增加注释
> > > 
> > > 
> > > 我用VS2022打开这个MD文件之后，是这个吗 ![屏幕截图 2024-09-03 185926](https://private-user-images.githubusercontent.com/180130297/363955330-6b57211d-3861-46ef-b0ae-e494ff713fda.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjE1ODYsIm5iZiI6MTcyNTM2MTI4NiwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU1MzMwLTZiNTcyMTFkLTM4NjEtNDZlZi1iMGFlLWU0OTRmZjcxM2ZkYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYWZjNGZjNDIxMGQyZThkNzQ1MzllZDM3ZDhkZGYwMmE2ZDhlNDg1NDE1N2MwZmVjZjgzOWM3NTE3MWQ5OWQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XqtwvzgW0TTRAKQRfE1AFvDQl0-TolS0HLvvzgjo7w8) ![屏幕截图 2024-09-03 185841](https://private-user-images.githubusercontent.com/180130297/363955474-36650e59-d705-46b9-a130-d80044742fa8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjE1ODYsIm5iZiI6MTcyNTM2MTI4NiwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU1NDc0LTM2NjUwZTU5LWQ3MDUtNDZiOS1hMTMwLWQ4MDA0NDc0MmZhOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MzUxYjZmZGVhZDVmOWJhNzM4MDJiMjI0MmY5ZThjZWIzMDI1YzJjNzI2Yjg2ZGU4MmZjZTdjODEwNTZjODZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.qyIJUq1QvZ5wg3k9mVMhAhzHKIW60VznpRRIrWQW0X8)
> > 
> > 
> > 这个是修改项目说明，比如增加贡献者名
> 
> ![屏幕截图 2024-09-03 190846](https://private-user-images.githubusercontent.com/180130297/363958005-8afa33fb-d256-4389-805f-30ee1b835705.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjIzODUsIm5iZiI6MTcyNTM2MjA4NSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU4MDA1LThhZmEzM2ZiLWQyNTYtNDM4OS04MDVmLTMwZWUxYjgzNTcwNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTE0NDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNGU3ODliMGI0Nzc1ZTBiM2IyMmQ2NzMwYjJiYTg5ZTFmZDI1ZDZjNmE0OWNmMjM1YjlkYzFmNTExMjFiMDc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JDS8kIeUtFRd8uV8Kme0lEpikINVGwIsHZ54lFwXnok) 不是要增加贡献者名吗，我把我的打上去了，emmm接下来该怎么增加上去。。。 ![屏幕截图 2024-09-03 191121](https://private-user-images.githubusercontent.com/180130297/363958801-5ca49d05-4415-4880-b101-b5407034accc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNjIzODUsIm5iZiI6MTcyNTM2MjA4NSwicGF0aCI6Ii8xODAxMzAyOTcvMzYzOTU4ODAxLTVjYTQ5ZDA1LTQ0MTUtNDg4MC1iMTAxLWI1NDA3MDM0YWNjYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QxMTE0NDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYmNlNmFhOTFlOWZmYTk1YjkwNGU2Nzc4NjM4MWRmN2RjMWNjNDUxODQ5MDFlYzExZDcwNjVhMDJmNGRkYzQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.OF4RGZpBn5GXVbPdvLOxBJ8qaOIe-9QY4BdHhKHfMfk)

参考贡献指南链接里的 https://zhuanlan.zhihu.com/p/153381521?utm_id=0

Candela-best, 2024-09-03:
> 看不懂最后一步，怎么把已改好的添加回去？

别看我这个，我的是误导的。。。

Candela-best, 2024-09-03:
> 参考贡献指南链接里的 https://zhuanlan.zhihu.com/p/153381521?utm_id=0

[image]
[image]
是这些都要吗，还是自己选

Candela-best, 2024-09-04:
> 参考贡献指南链接里的 https://zhuanlan.zhihu.com/p/153381521?utm_id=0

[image]
[image]
我上传了贡献者列表的文件，但是第二张截图中文件变化还是显示的是weather.cpp那个文件

closed 2025-02-09 by donghaiwang