OpenHUTB/hutb #67: 教程！内附详细步骤。不懂附截图来问。合作互赢，谢谢。

xiushuowang, 2024-09-04:
1.打开加速器
2.浏览器登录页面
3.进入老师那个页面
4.fork—> creating  fork
5.点击code ，复制你自己的https……网址（第七步要粘贴的）
6.回到电脑的主页，右键鼠标，点击显示更多选项，点击open git bash here
7.打入代码:
git clone +粘贴你的网址
8.待到页面出现一个文件夹，叫做carla_cpp时。点击进入该文件夹。
再点击README.md文档，用记事本打开，拉到最下层，打入你自己的网址，格式:
[ID]（https://github.com/ID） （你可以依葫芦画瓢，跟着别人的格式写）
9.更改成功后，点击保存。后叉掉记事本。
10.在Carla.cpp文件夹内，右键鼠标，进入显示更多选项，再次进入open Git bash here
11.先打入git add README.md代码
12.再打入 git status
13.再打入git commit -m "username"
14.再打入git push
15.缩小目前页面，等待代码反应，按操作登录
16.再次打入 git push
17.OK了
18登录你的GitHub
19.点击进入老师的页面（就是最最开始我们fork的那个页面）
20.当显示ahead的时候，证明我们可以了
21.屏幕有个contribute，点击这个，然后会出现一个绿色的按钮open pull request，点击进入，在主标题上写，申请贡献者列表，再提交就可以了。

leixing1, 2024-09-04:
非常感谢你，但是我打到git push那一步有问题，能否审查一下[
[image]
](url)

leixing1, 2024-09-04:
我打开github的网络是没有问题的

yyyyy1026, 2024-09-04:
按上面的步骤来的，为什么没有显示ahead

xiushuowang, 2024-09-04:
当中打入代码的时候，出错了。可以多试几次

liujue0, 2024-09-04:
电脑右键没有显示更多选项怎么办

xiushuowang, 2024-09-04:
没有更多选项是因为你安装了壁纸软件。建议关掉壁纸软件或者卸载

xiushuowang, 2024-09-04:
[image]

xiushuowang, 2024-09-04:
[image]

xiushuowang, 2024-09-04:
以上是各个跑成功的样子，各位同志可以参考

xiushuowang, 2024-09-04:
1.git add README.md第一个代码跑成功了，不会有显示
2.git status第二个代码跑成功了，会出现：modiffied：README.md
3.git commit -m "username"第三个代码跑成功了，会出现 1 file change， 1 insertion（+）等类似的东西
4.git push第四个代码成功后，会出现十行左右的代码，最后一行大概有个：  dev-> dev

liujue0, 2024-09-04:
更多选项里面没有open git bash here怎么办

xiushuowang, 2024-09-04:
那是因为你没有下载git 这个应用程序。在老师发布的那个页面，下拉有个git的下载链接

xiushuowang, 2024-09-04:
https://git-scm.com/downloads
这个就是git下载地址。要下二十几分钟，要有心理准备，我不知道怎么上传这个安装包，你们就自己下

xjc548, 2024-09-05:
大佬，我也来补充一个会出问题的情况，就是打代码的时候，出现fatal ：unable to access https......    前面这种情况的报错，可能是网不好，我也搞了好几次都没搞起，我把GitHub账号退了重新登录，就可以了。
希望能帮到大家

xuuu-1, 2024-09-05:
[image]
想问下这里怎么解决呀

xiushuowang, 2024-09-05:
> ![屏幕截图 2024-09-05 213106](https://private-user-images.githubusercontent.com/180184868/364797243-70bed580-9cbd-4ad2-8878-c85d8f0895f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NDQ0MTAsIm5iZiI6MTcyNTU0NDExMCwicGF0aCI6Ii8xODAxODQ4NjgvMzY0Nzk3MjQzLTcwYmVkNTgwLTljYmQtNGFkMi04ODc4LWM4NWQ4ZjA4OTVmMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNVQxMzQ4MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDk1ZmJhMTQwNmJmMDYxY2I5ZWEwOTZmNjYwNjgyM2FhODczOGJjOTJmMDE1ZTcyMmRjZjk2NmZiM2FiNWJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fidn7Tw2iFXDJIvhrwU1pUvEBMWRT2FhE0p49WTUPBs)想问下这里怎么解决呀

当前页面显示的是，你的仓库和学校仓库是一样的。如果你运行了那几个代码之后，还出现这个问题，建议翻翻评论区，你运行的代码没有运行成功，要重试。

xiushuowang, 2024-09-05:
> ![屏幕截图 2024-09-05 [...]

1.git add README.md 第一个代码跑成功了，不会有显示
2.git status第二个代码跑成功了，会出现：modiffied：README.md
3.git commit -m “username”第三个代码跑成功了，会出现 1 个文件更改，1 个插入（+）等类似的东西
4.git push第四个代码成功后，会出现十行左右的代码，最后一行大概有个： dev-> dev

xuuu-1, 2024-09-05:
那现在是要重试一次代码吗

xiushuowang, 2024-09-05:
> 那现在是要重试一次代码吗

是的。代码跑成功和没跑成功是不一样的情况。评论区有代码跑成功的简陋版介绍。实在不行发截图，一起讨论讨论。

xuuu-1, 2024-09-05:
[image]
这个地方报错了是为什么呀

xiushuowang, 2024-09-05:
> ![屏幕截图 2024-09-05 220625](https://private-user-images.githubusercontent.com/180184868/364810987-fbfa2943-3196-44d9-bcba-3af291403169.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NDU5OTIsIm5iZiI6MTcyNTU0NTY5MiwicGF0aCI6Ii8xODAxODQ4NjgvMzY0ODEwOTg3LWZiZmEyOTQzLTMxOTYtNDRkOS1iY2JhLTNhZjI5MTQwMzE2OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNVQxNDE0NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMGUwYmFmM2I2OWZjZjY3YTY3YWE4Y2ZmYjljYjBkYjVlOTNhZjZmOTQ3MmRmNjU4ZGFiZTliMmEyNzlmZjgzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ylP6VXLctepWyLSrXn9Lc5h5Cjxcdkmzzfe4OEJLjks) 这个地方报错了是为什么呀

上面显示了run.......to 
这就告诉我们，你把那两个代码复制下去再跑一遍。然后重新跑那四个代码就ok了。
git config --global user.email......
git config --global user.name......
跑这两个代码，代码我没写完，最好复制那个现成的。

xuuu-1, 2024-09-05:
[image]
这里xia显示一个减怎么回事呀

xuuu-1, 2024-09-05:
还想问下有冲突怎样同步来着，谢谢
[image]

xiushuowang, 2024-09-05:
> ![屏幕截图 2024-09-05 222701](https://private-user-images.githubusercontent.com/180184868/364819412-a7d3c4f8-72e0-41bc-a192-26d870fb7418.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1Nzc1NjUsIm5iZiI6MTcyNTU3NzI2NSwicGF0aCI6Ii8xODAxODQ4NjgvMzY0ODE5NDEyLWE3ZDNjNGY4LTcyZTAtNDFiYy1hMTkyLTI2ZDg3MGZiNzQxOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNVQyMzAxMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wOGNjZTlkNzA1NmE1ODUxMjRmNWFjODE1NzY1NGE0ZmZjMjI5YzFkNDM0Y2IwZWMzZWNhZGQyM2VlOGY5N2E5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Akq9mbCRdoPZMaqodl9rHqvfuPFsA4gM-UqdexFBzUY) 这里xia显示一个减怎么回事呀

说明你在更改文件的时候，删除了个什么东西。就有个-号

xiushuowang, 2024-09-05:
如果有冲突，在你自己的仓库里面。想要同步，有个sync fork，点这个就行，剩下的操作按要求去做。
同步完了，在重新走一遍流程即可。

52Herze, 2024-09-06:
> 1.git add README.md 第一个代码跑成功了，不会有显示 2.git status第二个代码跑成功了，会出现：modiffied：README.md 3.git commit -m “username”第三个代码跑成功了，会出现 1 个文件更改，1 个插入（+）等类似的东西 4.git push第四个代码成功后，会出现十行左右的代码，最后一行大概有个： dev-> dev
[image]
“username”需要具体填什么吗？

52Herze, 2024-09-06:
> 当中打入代码的时候，出错了。可以多试几次

[image]
没有ahead

Seavey0402, 2024-09-06:
第一遍试的时候没有显示ahead，第二遍到这里没有登录页面 
[image]

xiushuowang, 2024-09-06:
> 第一遍试的时候没有显示ahead，第二遍到这里没有登录页面 ![屏幕截图 2024-09-06 163950](https://private-user-images.githubusercontent.com/180534812/365096792-23fa901b-53ed-4e66-8450-876a4dac06a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MzU3NTksIm5iZiI6MTcyNTYzNTQ1OSwicGF0aCI6Ii8xODA1MzQ4MTIvMzY1MDk2NzkyLTIzZmE5MDFiLTUzZWQtNGU2Ni04NDUwLTg3NmE0ZGFjMDZhMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNlQxNTEwNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NjI5NWYzNzUyOTMxMThmYTY4Mzk5YzE5ZjY5YWFmNTE3MjMyMWNmYmUyNWMwNWViMGNhYWUxY2Q2OWZjY2M0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Nf3hjm4ex5U7cXjShGVbKn_E8nzzEZaW5B9Ux1I_pEU)

你这个是成功了，回到你自己的仓库页面，刷新即可。我想你现在应该解决这个问题了。

feng569150, 2024-09-08:
[image]
怎么都没办法打开，有人知道怎么办吗？

xiushuowang, 2024-09-08:
> ![屏幕截图 2024-09-08 150735](https://private-user-images.githubusercontent.com/180801212/365420307-98c30c46-4948-43f5-b386-26d44952ef46.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3ODg1MTgsIm5iZiI6MTcyNTc4ODIxOCwicGF0aCI6Ii8xODA4MDEyMTIvMzY1NDIwMzA3LTk4YzMwYzQ2LTQ5NDgtNDNmNS1iMzg2LTI2ZDQ0OTUyZWY0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwOFQwOTM2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYTdlYzE3YmM0Yzc4M2M5ZjA0Y2E5OTBiZWRjNDc3MjY5MzVmMGNlNjk2NjIxYTY1OTFkYTA5ZDI4MTcyN2NiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.q-9pvFBGsx3Ao3DZFIIfnMOb9VDTDDMb4ivOOHUkmPo)怎么都没办法打开，有人知道怎么办吗？

github账号退了重新登录，或者电脑重启，或者执行这个代码：git config --global http.sslVerify false

qqggxx, 2024-09-09:
第四步搞完显示403怎么回事

qqggxx, 2024-09-09:
卡在这怎么办
[image]

qqggxx, 2024-09-09:
第四个代码输入后显示403怎么回事

qqggxx, 2024-09-09:
第四个代码搞完后显示403是为什么

xiushuowang, 2024-09-09:
> 第四个代码搞完后显示403是为什么

发截图

qqggxx, 2024-09-09:
[image]

Mikecbk, 2024-09-09:
$ git commit -m “username”
Author identity unknown

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got '27495@▒▒֪▒▒▒▒▒▒.(none)')

这是输入第三个命令出现的，大佬帮帮我

xiushuowang, 2024-09-09:
> github账号退了重新登录，或者电脑重启

github账号退了重新登录，或者电脑重启

xiushuowang, 2024-09-09:
> git config --global user.email "[you@example.com](mailto:you@example.com)"
> git config --global user.name "Your Name"

在命令行跑这两个代码：
git config --global user.email "[you@example.com](mailto:you@example.com)"

git config --global user.name "Your Name"

xiushuowang, 2024-09-09:
> $ git commit -m “username” Author identity unknown
> 
> *** Please tell me who you are.
> 
> Run
> 
> git config --global user.email "[you@example.com](mailto:you@example.com)" git config --global user.name "Your Name"
> 
> to set your account's default identity. Omit --global to set the identity only in this repository.
> 
> fatal: unable to auto-detect email address (got '27495@▒▒֪▒▒▒▒▒▒.(none)')
> 
> 这是输入第三个命令出现的，大佬帮帮我

跑完再重新搞一次就行了

Mikecbk, 2024-09-09:
跑了很多次了

Mikecbk, 2024-09-09:
一开始没看到按流程跑完就是Everything up-to-date

Mikecbk, 2024-09-09:
感谢大佬，解决了！

qqggxx, 2024-09-09:
> > github账号退了重新登录，或者电脑重启
> 
> github账号退了重新登录，或者电脑重启

没用啊，我那个问题解决不了，还是一直显示403

jiameixiao, 2024-09-09:
原来我第三步有问题，后来输入俩个git config--global  user.----解决了  但是在第四步弄了几遍没弄好，后面重新在第三步的时候又出现了这个
[image]
然后进行第四步
[image]
用了这行代码还是出错了 但是会有下图出现
[image]
试了好几遍都是这样出现的结果，该怎么解决啊？

xiushuowang, 2024-09-10:
> 原来我第三步有问题，后来输入俩个git config--global user.----解决了 但是在第四步弄了几遍没弄好，后面重新在第三步的时候又出现了这个 ![屏幕截图 2024-09-09 234852](https://private-user-images.githubusercontent.com/180274580/365722325-768d7160-c33c-4e5e-8536-bc4d7a7121e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIyMzI1LTc2OGQ3MTYwLWMzM2MtNGU1ZS04NTM2LWJjNGQ3YTcxMjFlNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZDIwM2RhYjdjYjQ2ZDQ0ZDUzY2Q4NWMzZDE5ZDMzZTE4YzcyZDk0MDRmY2MyNjY5MjU1NWVlNzgwZjM5MzczJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SYVKn1p38Cl6cwGRJR_WeEpc8mG48AWvd30ASrO-gqM) 然后进行第四步 ![屏幕截图 2024-09-09 234933](https://private-user-images.githubusercontent.com/180274580/365722844-8be6b059-3371-4859-8d6c-7c5bee3750e4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIyODQ0LThiZTZiMDU5LTMzNzEtNDg1OS04ZDZjLTdjNWJlZTM3NTBlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NWFjMWQxZDU4ZmU4MjRlNDU4NDdkNmI2ZjU1ZWNjNjc5Zjg2NDQzMzYwYzRjMjRmNjg2OTI3OWMyYjY4NjQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1Ztg2W2FTwwIRMLnSFcRtj8t2yOeSlbVMvOVQJllv5Q) 用了这行代码还是出错了 但是会有下图出现 ![屏幕截图 2024-09-09 234944](https://private-user-images.githubusercontent.com/180274580/365723217-1c529087-b745-4a7a-b9b4-97ad69c33044.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIzMjE3LTFjNTI5MDg3LWI3NDUtNGE3YS1iOWI0LTk3YWQ2OWMzMzA0NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDczMzlkM2ZlY2E3YWU4NWQxZTkxM2FhNThjY2MxMGQ0OGRiZjhlNDJmZDk2ODQ4ZmM2YmViNjkyMGRlZGNmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.rYTknlBuMsJg7BtwWhpBskXG95EdmOfzploOF6RiyrI) 试了好几遍都是这样出现的结果，该怎么解决啊？

这一步是对的，就是要把GitHub和git绑定，点那个authorize git-ecosystem就可以了，你做到这一步就代表快成功了。

jiameixiao, 2024-09-10:
> 这一步是对的，就是要把GitHub和git绑定，点那个authorize git-ecosystem就可以了，你做到这一步就代表快成功了。

我点了那个authorize git-ecosystem 然后输入了密码  然后就出现了这个  那个可以点的authorize ,,,,页面也没有出现了
[image]

xiushuowang, 2024-09-10:
> > > 原来我第三步有问题，后来输入俩个git config--global user.----解决了 但是在第四步弄了几遍没弄好，后面重新在第三步的时候又出现了这个 ![屏幕截图 2024-09-09 234852](https://private-user-images.githubusercontent.com/180274580/365722325-768d7160-c33c-4e5e-8536-bc4d7a7121e6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIyMzI1LTc2OGQ3MTYwLWMzM2MtNGU1ZS04NTM2LWJjNGQ3YTcxMjFlNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZDIwM2RhYjdjYjQ2ZDQ0ZDUzY2Q4NWMzZDE5ZDMzZTE4YzcyZDk0MDRmY2MyNjY5MjU1NWVlNzgwZjM5MzczJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SYVKn1p38Cl6cwGRJR_WeEpc8mG48AWvd30ASrO-gqM) 然后进行第四步 ![屏幕截图 2024-09-09 234933](https://private-user-images.githubusercontent.com/180274580/365722844-8be6b059-3371-4859-8d6c-7c5bee3750e4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIyODQ0LThiZTZiMDU5LTMzNzEtNDg1OS04ZDZjLTdjNWJlZTM3NTBlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NWFjMWQxZDU4ZmU4MjRlNDU4NDdkNmI2ZjU1ZWNjNjc5Zjg2NDQzMzYwYzRjMjRmNjg2OTI3OWMyYjY4NjQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1Ztg2W2FTwwIRMLnSFcRtj8t2yOeSlbVMvOVQJllv5Q) 用了这行代码还是出错了 但是会有下图出现 ![屏幕截图 2024-09-09 234944](https://private-user-images.githubusercontent.com/180274580/365723217-1c529087-b745-4a7a-b9b4-97ad69c33044.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU5MzIwODQsIm5iZiI6MTcyNTkzMTc4NCwicGF0aCI6Ii8xODAyNzQ1ODAvMzY1NzIzMjE3LTFjNTI5MDg3LWI3NDUtNGE3YS1iOWI0LTk3YWQ2OWMzMzA0NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQwMTI5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDczMzlkM2ZlY2E3YWU4NWQxZTkxM2FhNThjY2MxMGQ0OGRiZjhlNDJmZDk2ODQ4ZmM2YmViNjkyMGRlZGNmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.rYTknlBuMsJg7BtwWhpBskXG95EdmOfzploOF6RiyrI) 试了好几遍都是这样出现的结果，该怎么解决啊？
> > 
> > 
> > 这一步是对的，就是要把GitHub和git绑定，点那个authorize git-ecosystem就可以了，你做到这一步就代表快成功了。
> 
> 我点了那个authorize git-ecosystem 然后输入了密码 然后就出现了这个 那个可以点的authorize ,,,,页面也没有出现了 ![屏幕截图 2024-09-10 235143](https://private-user-images.githubusercontent.com/180274580/366116556-3922db71-2267-43d2-8394-218a55b39a6e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYwMTAwNzcsIm5iZiI6MTcyNjAwOTc3NywicGF0aCI6Ii8xODAyNzQ1ODAvMzY2MTE2NTU2LTM5MjJkYjcxLTIyNjctNDNkMi04Mzk0LTIxOGE1NWIzOWE2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMFQyMzA5MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNjc2MWI0YTRhM2QyYmM1MzAxNmRlZGI2MWJkZTJkYjgwYjRkMzYxNzYxMTI3NjY0YmZiZThmMmVlODc2YjQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.pLoMHsVFZQqTr9ssOwzrFhUwBpsO9ok0gClZz-26vcg)

绑定了之后，再重新跑。记得要同步仓库，如果behind，就要重新sync fork。然后重新搞一遍就可以了。

jiameixiao, 2024-09-11:
> 绑定了之后，再重新跑。记得要同步仓库，如果behind，就要重新sync fork。然后重新搞一遍就可以了。

已经重新sync fork 
[image]
重新跑一遍还是出现
[image]

cenrain-m, 2024-09-11:
> 1.打开加速器 2.浏览器登录页面 3.进入老师那个页面 4.fork—> creating fork 5.点击code ，复制你自己的https……网址（第七步要粘贴的） 6.回到电脑的主页，右键鼠标，点击显示更多选项，点击open git bash here 7.打入代码: git clone +粘贴你的网址 8.待到页面出现一个文件夹，叫做carla_cpp时。点击进入该文件夹。 [...]

为什么第七百步代码运行失败

wanyZhu, 2024-09-11:
[image]
请问这是什么意思啊 第二步试了两次还是这样

xiushuowang, 2024-09-12:
> ![屏幕截图 2024-09-11 232807](https://private-user-images.githubusercontent.com/181249127/366523585-c11e3271-28d3-4d02-af71-741ad76a0866.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxMjU5MzgsIm5iZiI6MTcyNjEyNTYzOCwicGF0aCI6Ii8xODEyNDkxMjcvMzY2NTIzNTg1LWMxMWUzMjcxLTI4ZDMtNGQwMi1hZjcxLTc0MWFkNzZhMDg2Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxMlQwNzIwMzhaJlgtQW16LUV4cj1ietuGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjIxMmNmNjI3YTdkOTA5YTBhOGRlNzMzZDFkYWRhM2E5NTM2OTgxYjA5NDhmMmVmNDA1ZmU0Yjg2M2Y4OGVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lnNzobBU-5nfAgGmKviL-sfXmDPbA_i1XwYROPTt2r4) 请问这是什么意思啊 第二步试了两次还是这样

哪有问题？不是都对了吗?这个帖子有很多截图和纪录，可以参考一下。我没明白你的问题是什么

wanyZhu, 2024-09-12:
> 哪有问题？不是都对了吗?这个帖子有很多截图和纪录，可以参考一下。我没明白你的问题是什么

[image]
它整体是这样的，第三步没有出现那个加号，我没有看出来是哪里错了，而且具体进入老师的仓库的那一步ahead我也没有看见，这是啥问题啊

zhemuqi, 2024-09-15:
[image]

zhemuqi, 2024-09-15:
这个是为什么？重启完也不行

zhemuqi, 2024-09-15:
[image]

tushuguanL, 2024-10-16:
[image]
我运行了好几次了每次都卡在这个地方然后他就不动了是为什么呀

lingyanNB, 2024-11-18:
大神，我代码都跑成功了但没弹出登录界面是怎么回事，之有个登录界面，我准备授权的时候，点完那个authorize，那个页面进不去，所以我不知道绑定没，现在是这样，但没登录界面了
[image]

xiushuowang, 2024-11-18:
> 大神，我代码都跑成功了但没弹出登录界面是怎么回事，之有个登录界面，我准备授权的时候，点完那个authorize，那个页面进不去，所以我不知道绑定没，现在是这样，但没登录界面了
> [image]
> 

git页面不要全屏，拿小屏操作。没弹出界面，可以返回重新来一轮。但是要记得要sync同步，别behind了。

[22 more replies]

closed 2025-02-09 by donghaiwang