OpenHUTB/hutb #237: 如何使用TortoiseGit来更新文件

feixue31, 2024-09-12:


jun-72, 2024-09-13:
[image]
你按sshift+鼠标右键选择git同步 按我的图里面从左往右 从上往下依次点

minghao-lee, 2024-09-13:
直接拉取自己的同步的仓库就好了

feixue31, 2024-09-13:
> ![image](https://private-user-images.githubusercontent.com/180142120/367114005-ad9138f1-57c1-4755-accb-a3a9759eca65.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

我的页面怎么不一样
[image]

feixue31, 2024-09-13:
> ![image](https://private-user-images.githubusercontent.com/180142120/367114005-ad9138f1-57c1-4755-accb-a3a9759eca65.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

[image]
我的页面怎么不一样

jun-72, 2024-09-13:
你要配置一下TortoiseGit https://blog.csdn.net/xwnxwn/article/details/108694863

jun-72, 2024-09-13:
TortoiseGit是根据SSH加密来连接的 根据上面的来配置

feixue31, 2024-09-13:
好的，谢谢

feixue31, 2024-09-13:
我还想问一下就是使用了watt toolkithou加速后为什么就无法运行git pull来更新了？

jun-72, 2024-09-13:
> 我还想问一下就是使用了watt toolkithou加速后为什么就无法运行git pull来更新了？

这是计算机网络相关的内容 我没学过 要不问一问 @donghaiwang老师

donghaiwang, 2024-09-13:
是说小乌龟翻墙不了吗

[image]

在右键的小乌龟设置里

feixue31, 2024-09-13:
是不用小乌龟，直接在git bash here输入命令git pull的时候无法运行

donghaiwang, 2024-09-13:
> watt toolkithou

可能是watt toolkithou代理有问题，直接把watt toolkithou代理（或者系统代理）关了试试

closed 2025-02-09 by donghaiwang