OpenHUTB/hutb #255: 同样的代码在Dev c可以运行但在VS2019运行不了

jun-72, 2024-09-14:
[image]
[image]
如图

jun-72, 2024-09-14:
补图
[image]

jun-72, 2024-09-14:
补图
[image]

donghaiwang, 2024-09-14:
不要只贴图，还需要上传源代码文件和报错的日志信息。

jun-72, 2024-09-14:
> 不要只贴图，还需要上传源代码文件和报错的日志信息。

[image]
[分治排序算法.txt](https://github.com/user-attachments/files/17001922/default.txt)
github上传不了CPP文件 放在记事本里面了

closed 2025-02-09 by donghaiwang