OpenHUTB/hutb #264: 把文件从c盘换到d盘后Git打不开

kthalo, 2024-09-14:
[image]

jun-72, 2024-09-14:
[image]
看看你的环境变量有没有git

kthalo, 2024-09-15:
> ![image](https://private-user-images.githubusercontent.com/180142120/367522549-a8370380-8a25-4679-9ef8-54c89cae885f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzNjMyNjUsIm5iZiI6MTcyNjM2Mjk2NSwicGF0aCI6Ii8xODAxNDIxMjAvMzY3NTIyNTQ5LWE4MzcwMzgwLThhMjUtNDY3OS05ZWY4LTU0Yzg5Y2FlODg1Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxNVQwMTE2MDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYWJkYWExNjg1ZGQ5YzI5Njc2NjRiMjE5YWJmMDYzYjM4ZTc0ZDVmN2M1NzZhZDNkOGUxNzBhMGUxYmY3N2U3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UBR44vzuSH4_V0am4X3junvkOy6LnL0Q77uzktdUzDs) 看看你的环境变量有没有钱

[image]
有的，但它显示在c盘

closed 2025-02-09 by donghaiwang