OpenHUTB/hutb #273: 请问这怎么办

kthalo, 2024-09-15:
[image]

jun-72, 2024-09-15:
你没在carla_cpp这个文件夹里面git add .

kthalo, 2024-09-15:
> 你没在carla_cpp这个文件夹里面git add .

[image]
请问这又是什么问题呢

jun-72, 2024-09-15:
[image]
要么你把整个carla_cpp复制过去 因为你克隆过后他是直接连着你仓库的(如图只要连着仓库 左下角会有个绿色的√) 你的图中没有绿色的√ 说明没连接你仓库 重新clone一下吧

jun-72, 2024-09-15:
> > 你没在carla_cpp这个文件夹里面git add .
> 
> ![微信图片_20240915162857](https://private-user-images.githubusercontent.com/180262568/367568843-6e4b35ac-4774-4426-8f4b-bae1c60cd821.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzODk4ODQsIm5iZiI6MTcyNjM4OTU4NCwicGF0aCI6Ii8xODAyNjI1NjgvMzY3NTY4ODQzLTZlNGIzNWFjLTQ3NzQtNDQyNi04ZjRiLWJhZTFjNjBjZDgyMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxNVQwODM5NDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzA3YTEzYjc0YzQ4MzY5ZDdlN2UyMWEwZTA3MWUxZDlkYzg2NGI0YmFkOTA5MmNkYWYyMDJhYTY5NDAxMzJjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TBpXiOkhM73yCjQK4my33ef4dzoRJnBZTI5jTWFwr6I) 请问这又是什么问题呢

你不能自己建一个carla_cpp文件夹 然后把你进入你克隆过来carla_cpp的文件夹把里面的文件复制到你自己建的carla_cpp文件夹，你可以把整个克隆后的carla.cpp文件夹（带绿色的√复制过去，如图）
[image]

Yuuydz, 2024-09-16:
需要在文件所处文件夹中进行

closed 2025-02-09 by donghaiwang