OpenHUTB/hutb #795: 仓库克隆不了

ZH1024-Heng, 2024-10-17:
[image]

donghaiwang, 2024-10-17:
在TortoiseGit这个软件中也要设置代理

ZH1024-Heng, 2024-10-17:
[image]
在哪里设置代理？

donghaiwang, 2024-10-18:
> ![屏幕截图 2024-10-17 223231](https://private-user-images.githubusercontent.com/181330769/377511702-8e54c9a6-57ce-476b-a1b6-103da1b0db9b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkyNTEzMDcsIm5iZiI6MTcyOTI1MTAwNywicGF0aCI6Ii8xODEzMzA3NjkvMzc3NTExNzAyLThlNTRjOWE2LTU3Y2UtNDc2Yi1hMWI2LTEwM2RhMWIwZGI5Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxOFQxMTMwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zOTFjZDE2ZDVkNTY2NjhjNmFiNjNhZDRmOTllMmY5NWY4NmZjNmQ1NmZlODUwYWEzZjZlMzQwZDNiOTViYzlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.zTtCtG3P_YVfJr1Jxf5nHPfW9V2TjW7ttBbIrAfyxUI) 在哪里设置代理？

Network选项里

A-0-A-0-A, 2024-10-22:
也可以使用终端命令克隆

closed 2025-02-09 by donghaiwang