OpenHUTB/hutb #2141: 为什么insights里找不到自己的git账号？

lzg6667, 2024-12-20:


lzg6667, 2024-12-20:
为啥在贡献者里没有自己账号的数据呀

zero141271, 2024-12-20:
contributors中只会显示个

zero141271, 2024-12-20:
contributors中只会显示100个

lzg6667, 2024-12-20:
> contributors中只会显示100个

好的谢谢

zero141271, 2024-12-20:
可以在这里看到自己的提交结果
[image]
[image]
[image]

zero141271, 2024-12-20:
你可以这样查看自己的贡献
[image]
[image]
[image]

closed 2025-02-09 by donghaiwang