OpenHUTB/hutb #3249: 求求大佬指路——怎么运行配置脚本

sematic9527, 2025-07-19:
[image]

[image]
没有找到这个脚本

sematic9527, 2025-07-19:
> Setup.bat 是编译该仓库依赖 [虚幻引擎代码的脚本](http://127.0.0.1:8000/build_windows/#unreal-engine) ，不是这个仓库。
谢谢老师指导，但是我在您发的这个链接和windows构建文档中的链接中都点不进去

donghaiwang, 2025-07-19:
> > Setup.bat 是编译该仓库依赖 [虚幻引擎代码的脚本](http://127.0.0.1:8000/build_windows/#unreal-engine) ，不是这个仓库。
> > 谢谢老师指导，但是我在您发的这个链接和windows构建文档中的链接中都点不进去

抱歉，发的是本地服务的链接，这是公网服务的链接：https://openhutb.github.io/doc/build_windows/#unreal-engine

donghaiwang, 2025-07-19:
别客气

donghaiwang, 2025-07-19:
> <img alt="Image" width="1191" height="626" src="https://private-user-images.githubusercontent.com/180140974/468294126-44fb992f-2c80-4655-b10d-dee3c7bdaf0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI5MzY2NDgsIm5iZiI6MTc1MjkzNjM0OCwicGF0aCI6Ii8xODAxNDA5NzQvNDY4Mjk0MTI2LTQ0ZmI5OTJmLTJjODAtNDY1NS1iMTBkLWRlZTNjN2JkYWYwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxOVQxNDQ1NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MWZlYmIyZDVkOTBiZTU4ZjgzZTU1ZGU5ZDk2MDAzMDE5NDI5MTRmMGI4MzE4NjdhMGZkNzRlMDhlMmI5MDllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9._p2nCy6KNuzHDyj-2Wd57KKBD5d9fpJ525mjsBpiQQw">老师404，这个链接进不去

这是哪个链接

sematic9527, 2025-07-19:
[image]有个unreal-engine的蓝色字体

sematic9527, 2025-07-19:
> > <img alt="图像" width="1191" height="626" src="https://private-user-images.githubusercontent.com/180140974/468294126-44fb992f-2c80-4655-b10d-dee3c7bdaf0a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI5MzY2NDgsIm5iZiI6MTc1MjkzNjM0OCwicGF0aCI6Ii8xODAxNDA5NzQvNDY4Mjk0MTI2LTQ0ZmI5OTJmLTJjODAtNDY1NS1iMTBkLWRlZTNjN2JkYWYwYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNzE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDcxOVQxNDQ1NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MWZlYmIyZDVkOTBiZTU4ZjgzZTU1ZGU5ZDk2MDAzMDE5NDI5MTRmMGI4MzE4NjdhMGZkNzRlMDhlMmI5MDllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9._p2nCy6KNuzHDyj-2Wd57KKBD5d9fpJ525mjsBpiQQw">老师404，这个链接进不去
> 
> 这是哪个链接。引擎是私有仓库，需要进入社区才有权限查看，右键发你邀请链接了，看行不行。

老师我成了，我成了